RiftCat/vridge-api #3: Pitch and Roll axis problem in Desktop example app.

Desm0nt, 2017-05-03:
Good Day!

Desktop client example app have a problem with Pitch and Roll axis (or probably with all axis) for head and controller rotating (when real phone tracking work fine).  This problem similar to vridge freetrack implementation problem (report #11641). 

Because my English is bad, I can't fully discribe the problem. But I made a video with several demos (phone head rotation, head rotation emulation from app, and controller rotation emulation). I hope it can be useful.

[https://www.youtube.com/watch?v=e6rS-2XVjgk](url)

balukin, 2017-05-05:
Thanks for the video and detailed description! We'll check it next week and update you on our findings.

lonqq, 2017-05-11:
Hello,
We identified the problem and are currently working on it.

lonqq, 2017-05-17:
We've published an update with this fix on beta channel.

closed 2017-05-17 by lonqq