RiftCat/vridge-api #6: Remote API connection fail

balukin, 2017-06-16:
Effect:
Remote API connections fail. Only localhost connections work properly.

Problem source:
API server binds to localhost instead of all available IP addresses. 

Resolution:
Workaround for development purposes will be published here on Tuesday. Proper publicly available fix will follow later in June. Contact us if you need it released faster (for example, if you have product release relying on remote API coming soon).

mungewell, 2017-06-18:
Not dealing with remote connections, but early access to "HDMI/no-phone-needed" would really help my pyPSVR project development.

balukin, 2017-06-19:
@mungewell 
This will be released with 2.0 update which will be available for public testing this summer.

balukin, 2017-07-04:
Latest beta fixes it partially (because we didn't want to break backwards compatibility).

```EndpointAddress``` still contains ```localhost:port``` but you can replace "localhost" on client side with actual IP address of the server. Server listens on all available IP addresses now.

balukin, 2018-03-12:
Server now returns port only in endpoint response. Caller should now use remote IP address + given port to connect to data.

closed 2018-03-12 by balukin