RiftCat/vridge-api #7: Controller Packet - in python

mungewell, 2017-06-17:
Just like the Headset position, I am attempting to write python script to send Controller packets. Connection to the control API and sending data basically works, but I can't get 'orientation' working properly.

The following 'builds' a packet of 144 bytes total length:
```
controller_packet = Struct(
        Const(Int32ul, 2),              # Version
        Padded(4, Const(Byte, 1)),      # SendFullState
        Padded(4, Const(Byte, 0)),      # ControllerOrigin
        "id" / Int32ul,                 # ControllerId
        Const(Int32ul, 0),              # Status
        "orient" / Array(16, Float32l), # OrientationMatrix
        "count" / Int32ul,              # PacketNum
        "pressed" / Int64ul,            # ButtonPressed
        "touched" / Int64ul,            # ButtonTouched
        "axis" / Padded(40, Array(3, Float32l)),        # we only use 1st 3 of 10 floats
)
```

The controller appears, but at an 'odd' location, and rotation is very strange. It appears to rotate in one plane for 180' (of a 360' sweep) and then another plane for the 2nd 180'. Does the above look correct, it's very hard to decypher the wiki page?

[image]

mungewell, 2017-06-17:
A simplified version of what I am doing is as follows:

```
from pyrr import Quaternion, Matrix44, Vector3
<snip>

for swivel in range(0,360,30):
        # Direction that controller is pointing...?
        # [X, Z, Y]
        matrix1 = Matrix44.from_eulers([radians(0),radians(swivel),radians(0)])

        print("swivel:", swivel)
        print(matrix1)
        print(matrix1.reshape(16).tolist())
        print("---")

        controller1 = controller_packet.build(dict(
                id = 1111,
                orient = matrix1.reshape(16).tolist(),
                count = count,
                pressed = 1,
                touched = 0,
                axis = [0, 0, 0],
                ))
```

Output is attached. 
[test.txt](https://github.com/RiftCat/vridge-api/files/1082602/test.txt)

My (limited) understanding of Matrix math (quaterions and translation) is that a simple matrix like this should represent an spining object at the '(0,0,0,)' origin.
https://www.j3d.org/matrix_faq/matrfaq_latest.html

mungewell, 2017-06-18:
Think I might have got it working... took 4 bytes out of the start of the packet.

```
controller_packet = Struct(
        Const(Int32ul, 2),              # Version
        Padded(2, Const(Byte, 1)),      # SendFullState
        Padded(2, Const(Byte, 0)),      # ControllerOrigin

        "id" / Int32ul,                 # ControllerId
        Const(Int32ul, 0),              # Status
        "orient" / Array(16, Float32l), # OrientationMatrix
        "count" / Int32ul,              # PacketNum

        "pressed" / Int64ul,            # ButtonPressed
        "touched" / Int64ul,            # ButtonTouched
        "axis" / Padded(40, Array(3, Float32l)),        # we only use 1st 3 of 10 floats
)
```

And also need to 'transpose()' the vector to get the translations working.

```
        controller1 = controller_packet.build(dict(
                id = 1111,
                orient = matrix1.transpose().reshape(16).tolist(),
                count = count,
                pressed = 1,
                touched = 0,
                axis = [0, 0, 0],
                ))
```

mungewell, 2017-06-18:
working example code here:
https://github.com/mungewell/pyPSVR/blob/master/example3_vridgeapi/vridge_swivel_controllers.py

balukin, 2017-06-19:
I'm closing this issues since it seems to be resolved.

We'll be changing packets structs/serialization to make it less confusing. See https://github.com/RiftCat/vridge-api/issues/9.

closed 2017-06-19 by balukin