RiftCat/vridge-api #10: Controller ID/Type as seen by SteamVR

mungewell, 2017-06-25:
Hi,
Is there any way to control/configure what type of 'fake' controller we are presenting to SteamVR?

With my experiments (1) the controllers are rendered as Vive (Watchman) controllers in an empty construct, but currently are not rendered with SteamVR (were previously shown as Touch controllers). Can someone suggest a 'valid' ID value, perhaps they are somewhat coded?

For SteamVR Home I am 'stuck' as the configure panel shows a gamepad (rather than VR controllers) and says press 'Y' which does not exist. Pressing trigger brings us a pop-up, but triggering on the next button just closes rather than proceeding with the setup (to set head/heads type, etc).

1 https://github.com/mungewell/pyPSVR/tree/master/example3_vridgeapi

mungewell, 2017-06-25:
I found that 'SteamVR Home' can be launched seperately from 'Steam VR'. Disable it in the SteamVR config, then start (just) SteamVR... at this point I can start my script and see an empty construct with 2 two fave Vive controllers.

'SteamVR Home' can then be started from:

`c:\Program Files (x86)\Steam\steamapps\common\SteamVR\tools\steamvr_environments\game\bin\win64>.\steamtours.exe -vr`

Still doesn't get me Fake Vive controllers working, but it might be a step...
[image]

mungewell, 2017-06-25:
From 'vrsever.log':

> Sun Jun 25 2017 15:59:23.637 - oculus_vridge: New driver registered (serial VridgeControllerV02ID1) with status 1 
> Sun Jun 25 2017 15:59:23.639 - oculus_vridge: New driver registered (serial VridgeControllerV02ID2) with status 1 
> Sun Jun 25 2017 15:59:23.646 - oculus_vridge: Controller: VridgeControllerV02ID1 activated with id: 2
> Sun Jun 25 2017 15:59:23.646 - oculus_vridge: Controller: VridgeControllerV02ID2 activated with id: 3

mungewell, 2017-06-25:
"PseudoVive" does not appear to work with the current version of SteamVR, however it does hint on how to pretend to be a Vive Controller....

https://github.com/schellingb/PseudoVive/blob/master/Source/driver_2vive.cpp#L58

lonqq, 2017-06-26:
Hello, 
For now there is no way to change controller's model. This option might be available in the future, but for now there are some more important tasks. As for controllers IDs, in our API they are used only to distinguish controllers, and value does not affect how does controller works or looks. Keep in mind that different model would not change logic, it is being used for visual purpose only. 
We are not sure why SteamVR home does not accept virtual controllers, we do have similar issue with standard gamepads(Xbox One and Steam Controller), so we are not sure if problem is on our side.

closed 2017-06-26 by lonqq

mungewell, 2017-06-26:
For 'SteamVR Home' this is not just a visual problem. As noted above SteamVR Home is _unusable_ with the VRidge controllers as the 'Welcome Panel' (between the book shelves) is showing a gamepad (rather than a Vive controller), and a user is unable to pass the initial set-up of the Avatar/Hands.

It maybe possible to short-circuit this part of the 'SteamV Home' set-up with a pre-made Avatar/etc, but I don't know where this is store within the user settings/files.

Separately; For the DIYer it would be helpful to be able to fake any type of controller.

balukin, 2017-07-03:
There is a new problem with SteamVR home and controllers. They are completely invisible in current SteamVR version for some reason. I'm gonna open separate issue for that.