RiftCat/vridge-api #15: Async offset to be removed in v4

balukin, 2018-08-03:
Random offsets don't play well with new rotational smoothing and it doesn't seem like there's a real use case for it since asynchronous drift correction can lead to many problems.

benjanyan, 2018-08-05:
I've noticed that setting an offset in the beta version causes strange things to happen. As in the view ports seem to become smaller depending on the offset. Is that related to this?

If we still want to keep offset in check with V4, does this mean switching to the (potentially performance killing) synchronous headtracking API to correct drift? 

My own use case is using a Kinect to keep drift on an LG V30 in check (as well as provide headtracking). Otherwise I have to recenter fairly often.

balukin, 2018-08-06:
What you're seeing is reprojection correcting the rotation to original orientation. The higher the drift is, the lower the actually rendered viewport will be (because it becomes more and more out of sync). We left this unchanged in 2.2 because we didn't expect that headsets with reprojection (such as your, as it's Daydream-enabled) would require drift correction.

We discussed how to proceed with offset (both sync and async) because there were quite a lot of changes in 2.2.x versions related to positional tracking. In the end, it was extremely time-consuming to make the offset work in all cases (with API tracking, with Nolo tracking, with reprojection enabled, with controllers (AIO or API) and with any kind of combination of listed options). It becomes even more complex with controllers (some controllers can be recentered in their own environment, some controllers need to use our internal recentering (another offset), the complexity grows dramatically with each device's internal orientation system and recentering mechanism.

Eventually we decided to deprecate it because we can't assure that it will work in all cases. You already experience one of those problems (async offset + reprojection). 

Your options are:

1) The wild idea would be to use your external tracking system to detect when you're looking forward and call API's Recenter command every now and then. It might work but I haven't tested it and it might result in some problems. This API call results in the same action on the server side as pressing recenter hotkey.
2) We can build you an APK of 2.2.6 version that has reprojection disabled. This will significantly affect the smoothness in negative way but will still work with async offset. Let us know if you'd like us to build it. 
3) Stop using offsets until we figure how to bring it back universally to work with all combinations.

Sorry for the trouble. It can be solved with another layer of math and offset matrix recalculations, we just didn't want to spend a week on it because we weren't even sure if someone uses it. We don't have API access keys/analytics so we're sort of blind when it comes to knowing what is actually used by API developers.

We would like to fix it in v4/2.3 but I can't promise anything solid yet. Let me know if any of the above listed options is viable for you until then. We might come up with something else.

benjanyan, 2018-08-08:
Thanks for the comprehensive reply. I've been trying your first idea which sort of works although it's obviously not as flexible as using offset.

I wouldn't like to lose the reprojection as it is quite a nice feature. :) I may just wait and hope that you can come up with an idea to improve this.

I haven't got any other experience with other Daydream devices so I'm not sure if the drift I suffer from is specific to the LG V30. It's noticeably worse than the PS Move controllers.

balukin, 2018-08-08:
The Daydream is quite surprising, really. We didn't expect it as it never came up during our testing but it's always possible that we didn't run tests that were long enough to notice it drifting off few degrees.

We hope to have a solution for it because we need to refactor tracking systems a bit anyway. Currently the most limiting factor is API v1 (from VRidge 1.3 if I recall correctly) which is still compatible as Nolo desktop driver uses it so we can't drift off too much from original specs & tracking mixer. We'll figure something out.

balukin, 2018-10-05:
Progress update
I'm 99% sure that async offset will stay unchanged and won't have problems with reprojection in 2.3 update. We just need to run all the backwards compatibility tests before being 100% sure.

min050820, 2018-12-09:
Current on beta channel (2.3.9), Async offset does not work at all.
I provided yaw data but the camera did not move.

Tried settings:
task type = 201
data = [0, 1, 0], [0, 3.14, 0], [0, 100, 0], [100, 100, 100], [3.14, 3.14, 3.14], [1.57, 1.57, 1.57], [0, (0..10 linear), 0]

All did not work :(

Minor note: Using recenter for drift correction does not feel good because it snaps.

balukin, 2018-12-10:
Can you show me a code sample? 

I just tested both with new and older v3 API clients and it seems to work as expected.

Desktop example from this repo @ latest public commit (https://github.com/RiftCat/vridge-api/commit/0eef9ce41f8bd3edfde9f7d914c31ceb9a1d9cae)

min050820, 2018-12-13:
@balukin 
Okay, so it indeed was my fault. I'm terribly sorry.

balukin, 2019-02-01:
Closing. Discussed changes are now in effect and merged into stable VRidge channel and master branch of this repo.

Offset is now limited to yaw axis only. Older API calls will be converted and non-yaw data will be discarded but yaw offset should still be correctly applied without breaking reprojection. It will also correctly reorient 3rd party and our controllers.

closed 2019-02-01 by balukin