RiftCat/vridge-api #16: 3->6 dof controller reprojection

balukin, 2018-08-03:
Controller packet needs will include an option to let server know to reproject the 3DOF gyro data into 6DOF tracking, relative to head. 

Similarly to how Focus/Gear Vr/Go/Daydream controllers work.

BoomBrush, 2018-09-10:
Wondering if this has been implemented yet, and if the controller API docs could be updated to show how 6DOF tracking would work with a 3DOF controller. I'm having trouble understanding how to use a controller with 3DOF with the API

balukin, 2018-09-10:
It hasn't been updated yet. I'm currently refactoring a our tracking code to allow easier mixing of different tracking systems, paired with recentering, offsets, etc. 

When it's done it will allow API users to anchor their 3dof controllers to head. It will only require rotational data (and button states, of course) and will work like this: 

https://youtu.be/Kq629e_Gnew?t=19

It's not perfect but we played quite a lot of VR shooters and beat saber with this while were developing 3dof controller for HTC Vive Focus, Daydream, Oculus Go, and Gear VR controllers. 

It will use the same arm/hand/wrist model but it needed some extra work because there are many interactions between API, FreeTrack and phone tracking systems and we wanted to make sure that it works as good as it can with all the available options.

BoomBrush, 2018-09-12:
Ah, that makes sense as to why there is no documentation yet. Could the 3DOF inputs API allow for two controllers, or only one?

Thanks balukin

balukin, 2018-09-13:
You can send as many controllers as you want (each one with unique id) with a flag that will be evaluated per controller to 
1) Leave the controller data unchanged - used for controllers with 6dof tracking systems.
2) Remap the controller's pose to use left arm model according to its orientation relative to head.
3) Remap the controller's pose to use right arm model according to its orientation relative to head.

So there are no limitations at API level. You will be able to send one, two or even mix 6dof and 3dof controllers.

balukin, 2019-02-01:
The feature is now merged into master. Example API client implementation:

https://github.com/RiftCat/vridge-api/blob/master/src/csharp/VRE.Vridge.API.Client/Remotes/ControllerRemote.cs#L27

Summary:

Orientation matrix is now deprecated and should be null.

If position and orientation is provided, no remapping takes effect. If only orientation is provided and position is null, the 3->6DOF remapping will be in effect and position will be filled on the server side.

Applied on the controller level. 3 and 6 DOF controllers can be mixed.

closed 2019-02-01 by balukin