RiftCat/vridge-api #20: Set controller position in VRidge 2.2.7

HyroVitalyProtago, 2019-01-30:
Hello !

- Is the 3.1 api compatible with VRidge 2.2.7 ?
- Else, is there a way to set the controller position in the current api ?

PS: when I try to update VRidge in beta, my phone isn't detected anymore by usb, si if the 3.1 api is only compatible with the beta for now, I need to wait.

balukin, 2019-01-30:
3.1 API requires beta which is schedule to be merged into stable channel tomorrow. Beta is needed on both PC and Android side. Make sure both apps are in beta.

https://support.riftcat.com/hc/en-us/articles/360000572989-How-to-join-leave-the-VRidge-Beta-Update-channel-

closed 2019-01-30 by balukin

HyroVitalyProtago, 2019-01-30:
Congratulations for your responsiveness, I've forgot to update my android app to beta. Thanks a lot and good luck for the merge tomorrow !