RiftCat/vridge-api #22: How do

off99555, 2019-04-19:
I saw in the room setup of VRidge that it changes the text and input of the window. 
How do I do something similar to change the room setup window in steamVR?
Also how do I change the head tracking mechanism (I just want to adjust VRidge mechanism, not write from scratch)?
How do I access steam VR API like VRidge did?

Please concretely give an example if possible. I know C# in Unity and ZeroMQ but new to building C# project using NuGet or desktop app stuff.

balukin, 2019-04-23:
VRidge uses [OpenVR](https://github.com/ValveSoftware/openvr) driver to communicate with SteamVR. We expose small subset of SteamVR capabilities through our API to make it easier to extend VRidge own tracking (based on phone sensors) with DIY solutions and plugins. 

Could you describe what specifically would you like to alter in VRidge so I can point your in the right direction? Changing room setup process is not possible because it's all handled in SteamVR environment.

off99555, 2019-04-23:
OK. I want to use Vive tracker as the source for positional tracking. So I need to disable the rotation tracking from VRidge's gyro usage and use position/rotation (with offset) from tracker instead.
And also hide the tracker icon from SteamVR panel/dashboard so that people cannot access it alone. It should be merged with the headset.

balukin, 2019-04-23:
This is not something that can be done with VRidge-API alone. Supplying positional data to VRidge is easy. Rotational data is currently not allowed if your headset uses reprojection (but it's not a huge problem) because positional data from tracker with rotational data from phone should be good enough. 

The problem is that you need to interace with OpenVR anyway to extract data from the tracker. We haven't worked with Vive Trackers so you would need to investigate how to extract its data. What I can suggest to try instead is to use OpenVr tracking overrides. We haven't used it but theoretically you can try to override head position with Vive tracker position as explained [here](https://github.com/ValveSoftware/openvr/wiki/TrackingOverrides). In your case it would be something like:

```
"TrackingOverrides" : {
    "/devices/htc/{trackerSerialNumber}" : "/user/head"
},
```
However we've never done that so we are not sure if that would work as expected.

off99555, 2019-04-23:
OK. Cool. I'll investigate that. Thank you very much!

closed 2019-04-23 by off99555