RiftCat/vridge-api #23: How to access controller pose?

demonixis, 2019-04-30:
Hey,
I'm writting a Kinect driver. However I just want to submit the position of the hand, like for the head. For that I want to get hands pose to extract rotation.
Another idea could be to have a separate method to only send position.

balukin, 2019-05-07:
Let's clarify, you would like an API to list other controllers and query controller poses of other controllers registered in VRidge tracking system, is that correct?

demonixis, 2019-05-07:
Yes it is.

balukin, 2019-05-07:
Okay, I'll discuss this with the team and get back to this thread later when we decide. We're currently finishing up with iPhone tweaks for AMD systems so it might take a few more days.

demonixis, 2019-05-07:
That could be great guys! I've three drivers for RiftCat
- [Vridge.Kinect (one/360)](https://github.com/demonixis/Vridge.Kinect)
- [Vridge.NoloVR](https://github.com/demonixis/Vridge.NoloVR)
- [Vridge.3dRudder](https://github.com/demonixis/Vridge.3dRudder)

Allowing me to get those data could help a lot to improve the user experience. For instance I want to mix the rotations provided by the Phone controller app, as well as the Go controller, with the Kinect position. Having an API to get pose of connected devices would be awesome yes!

balukin, 2019-12-02:
This feature didn't make it into any of the updates because of the complexity of controller handling and transforms on the API server side. If more people request this feature, we will likely revisit this topic in the future.

A bit of a late update, I know and I'm sorry for that.