RiftCat/vridge-api #24: How can yaw offset be calculated from two quaternions?

AngelicosPhosphoros, 2019-05-31:
I try to adjust yaw using API to eliminate drift of headset sensors so I need to set YawOffset.

I got 2 quaternions: phone quaternion and target quaternion.

I know how to get quaternion delta (delta * phone_quat == my tracker quat). My tracker is attached directly on front side of headset.
But I don't understand how to get yaw offset from delta quaternion or from two quaternions.

Using yaw component of delta (I use this: https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles) is not working: image shaking every frame and often rotate in half round in both directions.

balukin, 2019-05-31:
Are you using ```RequestReadOnlyPhonePose``` or ```RequestReadOnlyPose```? The first one is marked as obsolete because using it as yaw offset reference would result in rotation skipping (because the pose already contains offset). Second one contains raw pose with offset information alongside but not included into raw pose.

The offset should be relative to raw phone data, not the pose that already includes the offset. This is the only scenario that comes to my mind that would result in rotation jumping wildly.

AngelicosPhosphoros, 2019-06-04:
Thanks a lot, using `RequestReadOnlyPose` was helpful.

However, there is a little tremor when I use yaw offsets.

balukin, 2019-06-05:
A) If you set the often too often (for example every other frame), it may result in a slight shakiness. If I recall correctly, reprojection may be out of sync for one frame if offset was applied recently. If you do this too often, it would result in a tremor like this. But in this case, offset shouldn't really be needed because all reprojection-enabled devices have very high quality rotational sensors of their own.

B) Instead of setting the offset to target offset value in one call, you may want to ease the transition by remembering the target offset in some variable in your process but moving towards the target offset vector in small increments, for example by 0.1 degree per frame only (you need to play around with numbers a little bit or configure it dynamically, depending how drifty the phone sensor is). Maybe this will help to keep the transition smooth.

closed 2019-12-02 by balukin