RiftCat/vridge-api #26: API rotation ignored due to  re-projection being enabled.

mungewell, 2020-11-12:
My python script is no longer able to update position. Console shows the message

```
Adapter 0 : NVIDIA GeForce GTX 960M
Adapter 1 : Intel(R) HD Graphics 520
03:41:33.577 [Info]: Creating GPU context.
03:41:33.659 [Info]: GPU context created on device NVIDIA GeForce GTX 960M / 76753
03:41:33.832 [Info]: API VRE.Vridge.API.Server.Services.Internal.WindowOutputService v1000 started for: internal
03:41:33.832 [Info]: API VRE.Vridge.API.Server.Services.Internal.InteropService v1000 started for: internal
03:41:33.832 [Info]: API VRE.Vridge.API.Server.Services.Internal.HeadTrackingServiceInternal v1000 started for: internal
03:41:43.557 [Info]: New pipe client connecting.
03:41:43.708 [Info]: Client connected in both directions.
03:41:43.708 [Info]: Client connected on thread 29.
03:41:46.796 [Info]: Starting capture: SteamVR @ 1920 x 1080
03:41:46.956 [Debug]: Starting vr source with params frAQBc8Wsa1xVPfvJcrgRYwTiizs2trQF69AzBlax3CF3EDNhm3soLBPh7 - -2147472958 / -2147473022 from process vrserver.
03:42:16.170 [Debug]: Discarding external tracking data because it tries to overwrite rotational data of reprojection-enabled device.
03:42:18.733 [Debug]: Discarding external tracking data because it tries to overwrite rotational data of reprojection-enabled device.
03:42:21.395 [Debug]: Discarding external tracking data because it tries to overwrite rotational data of reprojection-enabled device.
03:42:24.130 [Debug]: Discarding external tracking data because it tries to overwrite rotational data of reprojection-enabled device.
03:42:26.804 [Debug]: Discarding external tracking data because it tries to overwrite rotational data of reprojection-enabled device.
03:42:29.020 [Debug]: Discarding external tracking data because it tries to overwrite rotational data of reprojection-enabled device.
03:42:31.241 [Debug]: Discarding external tracking data because it tries to overwrite rotational data of reprojection-enabled device.
03:42:48.272 [Info]: Disconnecting external API overrides.
```

I have recently been playing with VRidge on the phone, and may have enabled this. Is there a way I can enable/disable reprojection via the API, or view keypresses in 'StreamView.exe'?

mungewell, 2020-11-12:
Found this section in 'vridge.cfg', setting false allows script to send rotations. Still if there is a way to set via API that would be good.

```
    "Name": "motorola one hyper",
    "RefreshRateHz": 60,
    "SupportsReprojection": false,
    "Type": 0,
    "Version": "2.7.4",
    "VersionCode": 20704
```

balukin, 2020-11-18:
This happens because API server remembers that most recently connected device was using reprojection. This is generally intended because if I recall correctly, device reprojection expects the render rotation to be matched with the pose captured on the device which has no way of working with API-driven head pose. 

Problem scenario occurs when you change from reprojection-enabled client to DIY solution because the "most recently connected device" field is not cleared so the engine still thinks that mobile reprojection is in effect. This can be worked around by flipping the flag that you posted but server fix would work better. I'll add it to our task board.